cesanta/mongoose #3125: "mdns-server" crashes on Windows

gvanem, 2025-05-03:
I wanted to try this new example program `mdns-server.exe` on Windows.
But it crashes since the `bind()` address `224.0.0.251:5353` returns `WSAEADDRNOTAVAIL`
and the sample program ignores a `c == NULL`. That's easy to fix:
```diff
--- a/tutorials/udp/mdns-server/main.c 2025-05-03 13:34:39
+++ b/tutorials/udp/mdns-server/main.c 2025-05-03 14:15:52
@@ -12,6 +12,9 @@

   // Desired name must NOT have any dots in it, nor a domain
   c = mg_mdns_listen(&mgr, "Mongoose");  // Start mDNS server
+  if (!c)
+    return 1;
+
   // if not using our built-in TCP/IP stack, pass the IP address you want to
```

But I fail to understand why Winsock2 won't accept this address in `bind()`.

BTW. what is `-DMG_ENABLE_MDNS=1` for? AFAICS, no code tests this?

scaprile, 2025-05-03:
> on Windows. But it crashes since the `bind()` address `224.0.0.251:5353` returns `WSAEADDRNOTAVAIL` and the sample program 

Please give us more data, our tests on Win worked fine. There shouldn't be an error returned, hence no test, this is not production code and should be as simple as possible.

> But I fail to understand why Winsock2 won't accept this address in `bind()`.

Exactly, it worked for us
 
> BTW. what is `-DMG_ENABLE_MDNS=1` for? AFAICS, no code tests this?

That's a leftover, initially we didn't default to enabling multicast due to many issues with small embedded systems, later we devised a cooler way but the OS code was not updated. I'll do it in the upcoming weeks, thanks for asking, I completely forgot about it.

gvanem, 2025-05-03:
> Please give us more data, our tests on Win worked fine. 

Seems both Chrome and Svchost is using this port; `netstat -4nau | grep 5353`:
```
udp   0.0.0.0:5353           -                         NONE         chrome.exe            16708
udp   0.0.0.0:5353           -                         NONE         svchost.exe            2608
udp   0.0.0.0:5353           -                         NONE         chrome.exe            15712
```

Hence this `WSAEADDRNOTAVAIL` in `mdns-server.exe` (but not in Chrome!?). 
`svchost.exe` / PID=2606 is the *DnsCache service* which is part of the *DNS Client* service. I'd rather not disable that.

closed 2025-05-08 by scaprile