cesanta/mongoose #3133: *** Panic *** Out of Momory

Electrolyt, 2025-05-10:
- My goal is: Connect to Amazon AWS per MQTT constant 
- My actions were: Connection to Amazon AWS. 
- My expectation was: hold connection. run continuesly
- The result I saw in the serial logging: 

b7b5   3 mongoose.c:4053:mg_connect     50 -1 mqtts://xxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com
b7cf   3 mongoose.c:299:dns_cb          50 xxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com is 52.209.194.25
b7d5   3 mongoose.c:5270:mg_connect_res 50 192.168.1.229:57204 -> 52.209.194.25:8883
b800   3 mongoose_glue.c:121:glue_mqtt_ 50 TLS enabled: yes

b97f   3 mongoose.c:11823:mg_tls_client certificate verification successful (RSA)

ba36   3 mongoose.c:4027:mg_close_conn  50 -1 closed

bb9d   3 mongoose.c:4053:mg_connect     51 -1 mqtts://xxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com
bbb7   3 mongoose.c:299:dns_cb          51 xxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com is 54.76.221.56
bbbe   3 mongoose.c:5270:mg_connect_res 51 192.168.1.229:57205 -> 54.76.221.56:8883
bbe7   3 mongoose_glue.c:121:glue_mqtt_ 51 TLS enabled: yes

*** PANIC ***

Out of memory

- My question is: Is there a bug in the source ?

# Environment
- mongoose version: 7.17
- Compiler/IDE and SDK: 
      cmake version 3.28.3
      The C compiler identification is GNU 13.2.1
      The CXX compiler identification is GNU 13.2.1
ZZZ version _X.Y_
- Target hardware/board: Seeduino XIAO-RP2350 (pico2)
- Connectivity chip/module: wiz850io 
- Target RTOS/OS (if applicable): -

scaprile, 2025-05-10:
I don't remember we having an explicit example doing that, so, whatever you did, is the case of your problem. That may also, of course, include trying to run something on a device with not enough memory for whatever you are trying to do.
Trace that panic message (which is not ours) and see where you run out of memory, determine why, then fix it.

closed 2025-05-10 by scaprile

Electrolyt, 2025-05-13:
thanks for the responce.
i changed the mcu hardware now and i am using a W5500-EVB-Pico board.
Same problem. i modified the build code minimal. it's 99,9% of your example.
i changed only the mqtt-adress, the mqtt-topic and the necessary TLS_CA CERTIFICATE.
but i got the same issue on this mcu.
it look like that mqtts triggers the issue.

scaprile, 2025-05-13:
> it look like that mqtts triggers the issue.

It may look like, but you need to know what it _is_. Use a debugger, find which function is running why the out of memory, who requests memory, how much, why, where.
We don't have Amazon accounts, we've used MQTT over TLS with other brokers and other boards, we run unit tests and a CI/CD pipeline. If you do have an issue that can be traced to our code, we'll fix it.