cesanta/mongoose #3144: Compiling Mongoos 7.18 throws a redundant redeclaration warning

jcorporation, 2025-05-21:
- My goal is: Update mongoose to newest release
- My actions were: Fetched the new source
- My expectation was: Compiles without warnings
- The result I saw: 

There is a rdeundant declaration warning.

```
myMPD/dist/mongoose/mongoose.c:8784:6: error: redundant redeclaration of ‘mg_multicast_add’ [-Werror=redundant-decls]
 8784 | void mg_multicast_add(struct mg_connection *c, char *ip);
      |      ^~~~~~~~~~~~~~~~
/home/juergen/projekte/myMPD/myMPD/dist/mongoose/mongoose.c:462:6: note: previous declaration of ‘mg_multicast_add’ with type ‘void(struct mg_connection *, char *)’
  462 | void mg_multicast_add(struct mg_connection *c, char *ip);
```

This can be fixed by removing the first line of below snippet.

```
void mg_multicast_add(struct mg_connection *c, char *ip);
void mg_multicast_add(struct mg_connection *c, char *ip) {
#if MG_ENABLE_RL
#error UNSUPPORTED
#elif MG_ENABLE_FREERTOS_TCP
  // TODO(): prvAllowIPPacketIPv4()
#else
  // lwIP, Unix, Windows, Zephyr(, AzureRTOS ?)
#if MG_ENABLE_LWIP && !LWIP_IGMP
  MG_ERROR(("LWIP_IGMP not defined, no multicast support"));
#else
  struct ip_mreq mreq;
  mreq.imr_multiaddr.s_addr = inet_addr(ip);
  mreq.imr_interface.s_addr = mg_htonl(INADDR_ANY);
  setsockopt(FD(c), IPPROTO_IP, IP_ADD_MEMBERSHIP, (char *) &mreq, sizeof(mreq));
#endif
#endif
}
```

- My question is: 

# Environment

- mongoose version: 7.18
- Compiler/IDE and SDK: gcc (Ubuntu 14.2.0-19ubuntu2) 14.2.0
- Target hardware/board: Linux t14 6.14.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Sun Apr  6 15:05:05 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

scaprile, 2025-05-21:
No warnings here, let me check.

jcorporation, 2025-05-21:
I compile it with:

```
"-DMG_ENABLE_DIRLIST=1"
"-DMG_ENABLE_SSI=0"
"-DMG_TLS=MG_TLS_OPENSSL"
```

scaprile, 2025-05-23:
@jcorporation That declaration is needed for other set of tests we do, it avoids having to include a header file and expose a function we don't intend to be user callable, the fact that it is not static is because we need it to be accessible among modules in this "unamalgamated" test (separate files instead of the amalgamated mongoose.c).
Newer gcc versions and even VC do not warn when the redundant declaration is really redundant, that is, it matches the previous one.
Please try adding `-Wno-redundant-decls` or whatever your compiler needs for that

closed 2025-05-23 by scaprile

jcorporation, 2025-05-24:
I use gcc 14.2 on Ubuntu 25.04, the toolchain should be rather fresh. But no Probleme, I will add the compiler flag.

Thanks for investigation!

scaprile, 2025-05-24:
Oh, sorry, I misread your data and took the gcc version as the Ubuntu version...
Looks like this is a very recent "feature", we'll see what we do (besides just adding that flag) once it hits our CI/CD