cesanta/mongoose #3160: Built-in TCP/IP: issue with setting local LAN DNS

scaprile, 2025-06-03:
mg_sendnsreq() assumes a connection can be immediately used after calling mg_connect (does not wait for a CONNECT event), so when "connecting" to a local DNS, ARP lookup is triggered but mg_send() is called right away, failing, and calling mg_error() to close the connection

OTOH, fixing this by just adding a CONNECT handler in dns_cb() will introduce the problem described in #3157, breaking the most common usage case.

actg, 2025-06-16:
I agree with you , I encounter the same problem as you.

scaprile, 2025-06-16:
@actg These are internal comments among Mongoose developers, we know what a "connection" means in Mongoose context and some of us have 25+ years of expertise in communications systems.
Of course you will have the same problem than I have, this is not a problem I have, see that I assigned the issue to myself ? I'm one of the developers, this is a verified Mongoose issue and we are working on it.
We will internally brainstorm and appropriately fix this. Thanks for your interest.

closed 2025-08-01 by cpq