cesanta/mongoose #3276: Configure TLS version (with openssl) i.e. allow calling SSL_set_options

mickem, 2025-09-18:
- My goal is: To allow end-users to configure the TLS version to use
- My actions were: Configure TLS with openssl
- My expectation was: That I could override tls options implementation so i can override them
- The result I saw: Is not possible to do
- My question is: Would it be possible to add an override to allow customizing tls options?

# Environment

- mongoose version: 7.19
- Compiler/IDE and SDK: MSVC++ version 141
- Target hardware/board: PC
- Connectivity chip/module: Intel?
- Target RTOS/OS (if applicable): N/A

A bit of background:
I have recently upgraded from an old version of mongoose where it was possible to override the tls options.
That seems no longer possible if I understand the code https://github.com/cesanta/mongoose/blob/master/mongoose.c#L14654-L14657 correctly.

I think it makes sense as some security conscious users might want to only allow tls 1.3.
It would be possible to either split `mg_tls_init` in two function one for creating the tls object and the other for configuring it allowing overriding the second function.
Another option would be to call an empty new function in `mg_tls_init` which can be overriden to set custom tls options.

scaprile, 2025-09-18:
Also, we support MbedTLS, OpenSSL, and WolfSSL, and won't introduce a function that is specific for OpenSSL. All necessary options are handled through the options structure.

scaprile, 2025-09-18:
Mickem, please describe your real life problem, not what you think is the solution to the code.
What _exactly_ is what you want to do in real life, and you can not do; so we can focus on how to provide that ?
Do you want to configure OpenSSL to force some TLS version, like 1.3 only ?

mickem, 2025-09-19:
Ohh, sorry...
Thats what I intended to say with: `My goal is: To allow end-users to configure the TLS version to use` where end-users here being the users of my compiled application where I embed the mongoose web server.
Sorry for my poor English language skills.

In other words I want to dynamically be able to switch supported tls version. 
So my application can have a command line option (not compile time option) that selects the supported TLS versions.
Something like:

```
my-app --supported-tls-version v1.3
...
my-app --supported-tls-version v1.1+
```

The reason for this is that some users use tools similar to https://github.com/testssl/testssl.sh where they validate the 
TLS version and other users instead require a slightly older version due to update cycles. 
So users of my compiled application will want to have different supported version of TLS.

This was possible in the older version I used before upgrading.

MGPhyo17, 2025-09-19:
// ...
#ifndef MG_CUSTOMIZE_TLS_OPTION
void override_tls_options(struct mg_tls *tls) {}
#endif

// and then call `override_tls_options` somewhere in  `mg_tls_init` 

  c->is_tls = 1;
  c->is_tls_hs = 1;
  MG_DEBUG(("%lu SSL %s OK", c->id, c->is_accepted ? "accept" : "client"));
  override_tls_options(tls);
  return;