cesanta/mongoose #3278: Sending HTTPS request has memory leak

general252, 2025-09-19:
- My goal is: Sending HTTPS request has memory leak
- My actions were: Send HTTPS request
- My expectation was: memory leak
- The result I saw: _provide logs or description_
- My question is: `mg_tls_free` need `mbedtls_psa_crypto_free()`

# Environment

- mongoose version:  7.19 
- Compiler/IDE and SDK: gcc version 11.4.0
- Target hardware/board: PC
- Connectivity chip/module: Intel
- Target RTOS/OS (if applicable): N/A

`mg_tls_free` need `mbedtls_psa_crypto_free()`

```C
void mg_tls_free(struct mg_connection *c) {
  struct mg_tls *tls = (struct mg_tls *) c->tls;
  if (tls != NULL) {
#if defined(MBEDTLS_VERSION_NUMBER) && MBEDTLS_VERSION_NUMBER >= 0x03000000 && \
    defined(MBEDTLS_PSA_CRYPTO_C)
      mbedtls_psa_crypto_free();
#endif

    mbedtls_ssl_free(&tls->ssl);
    mbedtls_pk_free(&tls->pk);
    mbedtls_x509_crt_free(&tls->ca);
    mbedtls_x509_crt_free(&tls->cert);
    mbedtls_ssl_config_free(&tls->conf);
#ifdef MBEDTLS_SSL_SESSION_TICKETS
    mbedtls_ssl_ticket_free(&tls->ticket);
#endif
    mg_free(tls);
    c->tls = NULL;
  }
}
```

```bash
# valgrind --tool=memcheck --leak-check=full --show-leak-kinds=all ./a.out
==28248== Memcheck, a memory error detector
==28248== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==28248== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==28248== Command: ./a.out
==28248==
CJSON version: 1.7.12
mongoose version: 7.19
mbedtls version: 3.6.4
629933 2 auth_client.c:62:_mg_http_call ----
62994c 2 auth_client.c:76:_mg_http_call --MG_EV_CONNECT
629c00 2 auth_client.c:87:_mg_http_call --MG_EV_HTTP_HDRS
629c07 2 auth_client.c:87:_mg_http_call --MG_EV_HTTP_HDRS
200 3360
629c10 2 auth_client.c:106:_mg_http_cal --MG_EV_HTTP_MSG
629c15 2 auth_client.c:125:_mg_http_cal mg_tls_free
629c25 2 auth_client.c:128:_mg_http_cal --MG_EV_CLOSE

==28248==
==28248== HEAP SUMMARY:
==28248==     in use at exit: 2,904 bytes in 3 blocks
==28248==   total heap usage: 10,379 allocs, 10,376 frees, 503,972 bytes allocated
==28248==
==28248== 216 bytes in 1 blocks are still reachable in loss record 1 of 3
==28248==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==28248==    by 0x1AA037: mbedtls_md_setup (md.c:481)
==28248==    by 0x1A38A2: mbedtls_entropy_func (entropy.c:340)
==28248==    by 0x191FBB: mbedtls_ctr_drbg_reseed_internal (ctr_drbg.c:469)
==28248==    by 0x19226C: mbedtls_ctr_drbg_seed (ctr_drbg.c:583)
==28248==    by 0x1B69FD: mbedtls_psa_drbg_seed (psa_crypto_random_impl.h:117)
==28248==    by 0x1C343E: mbedtls_psa_random_seed (psa_crypto.c:8019)
==28248==    by 0x1C3D86: mbedtls_psa_crypto_init_subsystem (psa_crypto.c:8480)
==28248==    by 0x1C3E26: psa_crypto_init (psa_crypto.c:8558)
==28248==    by 0x135B27: mg_tls_init (mongoose.c:14179)
==28248==    by 0x112B0D: _mg_http_callback (auth_client.c:68)
==28248==    by 0x11E6E4: mg_call (mongoose.c:499)
==28248==
==28248== 896 bytes in 1 blocks are still reachable in loss record 2 of 3
==28248==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==28248==    by 0x1CCC67: psa_allocate_volatile_key_slot (psa_crypto_slot_management.c:516)
==28248==    by 0x1CCF40: psa_reserve_free_key_slot (psa_crypto_slot_management.c:609)
==28248==    by 0x1B9276: psa_start_key_creation (psa_crypto.c:1801)
==28248==    by 0x1C3851: psa_generate_key_custom (psa_crypto.c:8220)
==28248==    by 0x1C3B94: psa_generate_key (psa_crypto.c:8291)
==28248==    by 0x177E68: mbedtls_ssl_tls13_generate_and_write_xxdh_key_exchange (ssl_tls13_generic.c:1586)
==28248==    by 0x16EB2A: ssl_tls13_write_key_share_ext (ssl_tls13_client.c:318)
==28248==    by 0x170B66: mbedtls_ssl_tls13_write_client_hello_exts (ssl_tls13_client.c:1180)
==28248==    by 0x17A77B: ssl_write_client_hello_body (ssl_client.c:598)
==28248==    by 0x17AF71: mbedtls_ssl_write_client_hello (ssl_client.c:929)
==28248==    by 0x14AB23: mbedtls_ssl_handshake_step (ssl_tls.c:4596)
==28248==
==28248== 1,792 bytes in 1 blocks are still reachable in loss record 3 of 3
==28248==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==28248==    by 0x1CCA7D: psa_initialize_key_slots (psa_crypto_slot_management.c:434)
==28248==    by 0x1C3D2B: mbedtls_psa_crypto_init_subsystem (psa_crypto.c:8441)
==28248==    by 0x1C3E13: psa_crypto_init (psa_crypto.c:8553)
==28248==    by 0x135B27: mg_tls_init (mongoose.c:14179)
==28248==    by 0x112B0D: _mg_http_callback (auth_client.c:68)
==28248==    by 0x11E6E4: mg_call (mongoose.c:499)
==28248==    by 0x132E57: connect_conn (mongoose.c:9023)
==28248==    by 0x134482: mg_mgr_poll (mongoose.c:9407)
==28248==    by 0x113284: _fn_poll (auth_client.c:200)
==28248==    by 0x11C892: _test_http (main.c:123)
==28248==    by 0x11C9BB: main (main.c:160)
==28248==
==28248== LEAK SUMMARY:
==28248==    definitely lost: 0 bytes in 0 blocks
==28248==    indirectly lost: 0 bytes in 0 blocks
==28248==      possibly lost: 0 bytes in 0 blocks
==28248==    still reachable: 2,904 bytes in 3 blocks
==28248==         suppressed: 0 bytes in 0 blocks
==28248==
==28248== For lists of detected and suppressed errors, rerun with: -s
==28248== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

scaprile, 2025-09-19:
We test with ASAN and Valgrind in a CI/CD pipeline, and we don't see that.
Please don't post our code, if you need to refer to it, then paste a oermalink to it; let us see yours.
Please provide which TLS library are you using and its version, and a small piece of code to reproduce your issue.
I guess your "PC" has an OS, doesn't it ? Please fill all fields in the issue template, it is there for a reason: to save yours and our time; thank you.

general252, 2025-09-22:
> We test with ASAN and Valgrind in a CI/CD pipeline, and we don't see that. Please don't post our code, if you need to refer to it, then paste a oermalink to it; let us see yours. Please provide [...]
My issues has been updated

general252, 2025-09-22:
mbedtls programs example.  on free call `mbedtls_psa_crypto_free()`
https://github.com/Mbed-TLS/mbedtls/blob/v3.6.4/programs/ssl/dtls_client.c#L94-L109
https://github.com/Mbed-TLS/mbedtls/blob/v3.6.4/programs/ssl/dtls_client.c#L321-L331

scaprile, 2025-09-22:
We test under Ubuntu on several CI/CD pipelines, we run our test code and we don't have memory leaks with our suggested usage.
Your links are to some part of some library and the name is DTLS. We don't support DTLS, your report is about HTTP.
The linked code shows some library usage, not how you use Mongoose.

I'm closing this, if you actually have a problem when using Mongoose with the provided documentation and examples, provide us with pertinent information in order to reproduce and fix the issue. We don't inspect your code for errors, please _attach_ the minimum piece of code that can reproduce the issue. If you _post_ code, please correctly format it so Github shows it as code. We'll then re-open and check.

Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).

closed 2025-09-22 by scaprile